jbanaszczyk/pms5003 #3: Compatibility with Arduino Mega 2560

Roberto6969, 2017-04-10:
Do you plan to make library compatible with Arduino Mega 2560? If i use it on Mega i cannot retrieve data from the sensor (case Pms5003::noData:). Is there any special requirements for Mega? Thank you!

jbanaszczyk, 2017-04-10:
About the plans: yes, probably I'll test it with Mega in the not so far future. 
My priority is to clean-up the first version, including docs.

How did you connect sensor and Arduino? Have you used logic converter?

Roberto6969, 2017-04-10:
I have used active logic converter. I made four tests:
1. Arduino UNO with PMS - works fine
2. Arduino Mega 2560 with PMS - no data read
3. Arduino UNO with Dragino v1.3 LoRa shield with PMS (exactly the same connections as in test 1) - no data read; this is quite strange because between Arduino and PMS is only the shield
4. Arduino Mega 2560 with Dragino v1.3 LoRa shield with PMS (again the same connections) - no data read

In all four test i used the same minimum sketch to read the PMS senzor.  If you need any more informations please let me know.

jbanaszczyk, 2017-04-12:
I'll try to switch my board (Arduino Uno Plus) to 5V and use converter - probably during the weekend.

According to https://codebender.cc/example/AltSoftSerial/Test#Test.ino have you used pins 46,48 ?

Roberto6969, 2017-04-13:
Good question and the simple answer is: no. I missed this. I have tested with Arduino MEGA 2560, BMP280 on I2C, PMS5003 and Dragino LoRa shield v1.3 with LMIC library and it works. The only problem is sometimes i get the error "sumError" but i believe this is due to coexisting code which needs some optimization. Anyway, in what cases could this error emerge?

Roberto6969, 2017-04-28:
After tests i found out where is the problem. If the main code to read PMS is within void loop () than it is OK and there is no error. But if the loop () just call separate function with exactly the same code to read PMS, than the errors becoming more an more often.  Is there any limitations or any other conditions for function to read PMS data?

jbanaszczyk, 2017-04-28:
Please share the code causing errors?

Roberto6969, 2017-04-29:
Here it is. If the content of readPMS() is inside of void loop() than works without any error :

/*
   pin 4 on D9
   pin 5 on D8
   no pullup resistors, RESET and SET are not connected
   SET = LOW means sensor sleep mode
*/
#include <Arduino.h>
#include <pms.h>
#include <LowPower.h>
//#include <AltSoftSerial.h>

Pms5003 pms;

////////////////////////////////////////

void setup(void) {
  Serial.begin(115200);
  while (!Serial) {};
  Serial.println("PMS5003");

  pms.begin();
  pms.waitForData(Pms5003::wakeupTime);
  pms.write(Pms5003::cmdModeActive);
}

////////////////////////////////////////

auto lastRead = millis();

void loop(void) {
  Serial.println("Main loop");
  readPMS();
  delay(5000);
}

void readPMS () {
  const auto n = Pms5003::Reserved;
  Pms5003::pmsData data[n];
  Pms5003::PmsStatus status = pms.read(data, n);

  switch (status) {
    case Pms5003::OK:
      {
        Serial.println("_________________");
        auto newRead = millis();
        Serial.print("Wait time ");
        Serial.println(newRead - lastRead);
        lastRead = newRead;

        // For loop starts from 3
        // Skip the first three data (PM1dot0CF1, PM2dot5CF1, PM10CF1)
        for (size_t i = Pms5003::PM1dot0; i < n; ++i) {
          Serial.print(data[i]);
          Serial.print("\t");
          Serial.print(Pms5003::dataNames[i]);
          Serial.print(" [");
          Serial.print(Pms5003::metrics[i]);
          Serial.print("]");
          Serial.println();
        }
        break;

      }
    case Pms5003::noData:
      //Serial.println("Did not get any data from the sensor!!!");
      break;
    default:
      Serial.println("______________ E R R O R !!! _______________");
      Serial.println(Pms5003::errorMsg[status]);
  };
}

closed 2018-04-23 by jbanaszczyk

CAPlevel, 2019-12-01:
Did you have a solution for this?
I’m using the library and works great with Esp8266 (wemos d1 mini), I moved the simple test out of main loop with a call and it fail several time with a SumError.

CAPlevel, 2019-12-01:
This is what I found. even without moving the code out of look.
Using the Simple test Adding a 5 seconds delay just before the LOOP() end, I get the sum error almost immediately after the first read., then more sumerrors..

here are the last pieces of code with the simple test that ALWAYS create the error, how can I fix it, please?

		case Pmsx003::noData:
			break;
		default:
			Serial.println("_________________");
			Serial.println(Pmsx003::errorMsg[status]);
	};
  delay(5000);
}