jbanaszczyk/pms5003 #4: Problem Including AltSoftSerial.h

Legend2015, 2017-04-28:
Hello, I am trying to use this library for my PMS sensors and have run into a problem. I am using the modified AltSoftSerial Library as suggested. But when i try to compile i am getting this error. 

> Arduino: 1.6.5 (Windows 8.1), Board: "Arduino/Genuino Uno"
>
>Build options changed, rebuilding all
>
>In file included from C:\Program Files (x86)\Arduino\libraries\pms5003-master\src/pms.h:7:0,
 >                from PMS5003_1.ino:3:
>C:\Program Files (x86)\Arduino\libraries\pms5003-master\src/pmsConfig.h:37:27: fatal error: >AltSoftSerial.h: No such file or directory
 >#include <AltSoftSerial.h>
>                           ^
>compilation terminated.
>Error compiling.

  Could you please help me as to what might be going wrong?

Thank you so much

jbanaszczyk, 2017-04-28:
AltSoftSerial is not installed.
Follow the link: https://www.arduino.cc/en/Guide/Libraries

closed 2017-04-28 by jbanaszczyk

Legend2015, 2017-04-29:
Hi,

I can see AltSoftSerial installed when i go Sketch -> Include Library under the Contributed Libraries section. Also, when i manually add "#include <AltSoftSerial.h>" in the Simple01.ino it gives mea different set of errors like "'wakeupTime' is not a member of 'Pms5003'". The same error comes up after i manually changed "#include <AltSoftSerial.h>" to "#include <../../AltSoftSerial/AltSoftSerial.h> in pmsConfig.h. That tells me it is able to find the AltSoftSerial.h this way. The full text of the error is as follows:

`Arduino: 1.6.5 (Mac OS X), Board: "Arduino/Genuino Uno"

In file included from /Applications/Arduino.app/Contents/Java/libraries/pms5003/src/pms.h:6:0,
                 from Simple01.ino:3:
/Applications/Arduino.app/Contents/Java/libraries/pms5003/src/tribool.h:19:1: error: 'constexpr' does not name a type
 constexpr inline bool unknown(tribool arg, unknown_t dummy = unknown_t());
 ^
/Applications/Arduino.app/Contents/Java/libraries/pms5003/src/tribool.h:19:1: note: C++11 'constexpr' only available with -std=c++11 or -std=gnu++11
/Applications/Arduino.app/Contents/Java/libraries/pms5003/src/tribool.h:25:2: error: 'constexpr' does not name a type
  constexpr tribool() : value(unknown_v) {}
  ^
/Applications/Arduino.app/Contents/Java/libraries/pms5003/src/tribool.h:25:2: note: C++11 'constexpr' only available with -std=c++11 or -std=gnu++11
/Applications/Arduino.app/Contents/Java/libraries/pms5003/src/tribool.h:26:2: error: 'constexpr' does not name a type
  constexpr tribool(bool v) : value(v ? true_v : false_v) {};
  ^
/Applications/Arduino.app/Contents/Java/libraries/pms5003/src/tribool.h:26:2: note: C++11 'constexpr' only available with -std=c++11 or -std=gnu++11
/Applications/Arduino.app/Contents/Java/libraries/pms5003/src/tribool.h:27:2: error: 'constexpr' does not name a type
  constexpr tribool(unknown_keyword_t) : value(unknown_v) {}
  ^
/Applications/Arduino.app/Contents/Java/libraries/pms5003/src/tribool.h:27:2: note: C++11 'constexpr' only available with -std=c++11 or -std=gnu++11
/Applications/Arduino.app/Contents/Java/libraries/pms5003/src/tribool.h:28:2: error: 'constexpr' does not name a type
  constexpr inline operator bool() const {
  ^
/Applications/Arduino.app/Contents/Java/libraries/pms5003/src/tribool.h:28:2: note: C++11 'constexpr' only available with -std=c++11 or -std=gnu++11
/Applications/Arduino.app/Contents/Java/libraries/pms5003/src/tribool.h:32:2: error: 'constexpr' does not name a type
  constexpr inline tribool operator!() const {
  ^
/Applications/Arduino.app/Contents/Java/libraries/pms5003/src/tribool.h:32:2: note: C++11 'constexpr' only available with -std=c++11 or -std=gnu++11
/Applications/Arduino.app/Contents/Java/libraries/pms5003/src/tribool.h:41:1: error: 'constexpr' does not name a type
 constexpr inline tribool operator&&(tribool lhs, tribool rhs) {
 ^
/Applications/Arduino.app/Contents/Java/libraries/pms5003/src/tribool.h:41:1: note: C++11 'constexpr' only available with -std=c++11 or -std=gnu++11
/Applications/Arduino.app/Contents/Java/libraries/pms5003/src/tribool.h:48:1: error: 'constexpr' does not name a type
 constexpr inline tribool operator&&(tribool lhs, bool rhs) { return rhs ? lhs : tribool(false); }
 ^
/Applications/Arduino.app/Contents/Java/libraries/pms5003/src/tribool.h:48:1: note: C++11 'constexpr' only available with -std=c++11 or -std=gnu++11
/Applications/Arduino.app/Contents/Java/libraries/pms5003/src/tribool.h:49:1: error: 'constexpr' does not name a type
 constexpr inline tribool operator&&(bool lhs, tribool rhs) { return lhs ? rhs : tribool(false); }
 ^
/Applications/Arduino.app/Contents/Java/libraries/pms5003/src/tribool.h:49:1: note: C++11 'constexpr' only available with -std=c++11 or -std=gnu++11
/Applications/Arduino.app/Contents/Java/libraries/pms5003/src/tribool.h:50:1: error: 'constexpr' does not name a type
 constexpr inline tribool operator&&(unknown_keyword_t, tribool lhs) { return !lhs ? tribool(false) : tribool(unknown); }
 ^
/Applications/Arduino.app/Contents/Java/libraries/pms5003/src/tribool.h:50:1: note: C++11 'constexpr' only available with -std=c++11 or -std=gnu++11
/Applications/Arduino.app/Contents/Java/libraries/pms5003/src/tribool.h:51:1: error: 'constexpr' does not name a type
 constexpr inline tribool operator&&(tribool lhs, unknown_keyword_t) { return !lhs ? tribool(false) : tribool(unknown); }
 ^
/Applications/Arduino.app/Contents/Java/libraries/pms5003/src/tribool.h:51:1: note: C++11 'constexpr' only available with -std=c++11 or -std=gnu++11
/Applications/Arduino.app/Contents/Java/libraries/pms5003/src/tribool.h:53:1: error: 'constexpr' does not name a type
 constexpr inline tribool operator||(tribool lhs, tribool rhs) {
 ^
/Applications/Arduino.app/Contents/Java/libraries/pms5003/src/tribool.h:53:1: note: C++11 'constexpr' only available with -std=c++11 or -std=gnu++11
/Applications/Arduino.app/Contents/Java/libraries/pms5003/src/tribool.h:60:1: error: 'constexpr' does not name a type
 constexpr inline tribool operator||(tribool lhs, bool rhs) { return rhs ? tribool(true) : lhs; }
 ^
/Applications/Arduino.app/Contents/Java/libraries/pms5003/src/tribool.h:60:1: note: C++11 'constexpr' only available with -std=c++11 or -std=gnu++11
/Applications/Arduino.app/Contents/Java/libraries/pms5003/src/tribool.h:61:1: error: 'constexpr' does not name a type
 constexpr inline tribool operator||(bool lhs, tribool rhs) { return lhs ? tribool(true) : rhs; }
 ^
[... cut ...]
Simple01:26: error: 'n' was not declared in this scope
Simple01:28: error: 'data' was not declared in this scope
Simple01:34: error: 'newRead' does not name a type
Simple01:36: error: 'newRead' was not declared in this scope
Simple01:36: error: 'lastRead' was not declared in this scope
'wakeupTime' is not a member of 'Pms5003'

  This report would have more information with
  "Show verbose output during compilation"
  enabled in File > Preferences.
`
I tried it in both my Windows 10 laptop and MacBook Pro, with same errors. :(

jbanaszczyk, 2017-05-04:
You use very old Arduino, without support for C++11. Please upgrade.