jbanaszczyk/pms5003 #8: Sleep/deep sleep/wakeuptime

maddhin, 2017-09-23:
Hi, 
I love the library and try to put together a script that allows to have the PMS sleep for a while. I simply want to measure PM only every - say - 10min and then go to (deep) sleep. 

But I am struggling to realize this. At the moment I am working with delay() but that has the effect, that "something times out" and I do not get ANY readings.

I want to realize that the ESP8266 wakes up, waits an appropriate time so the sensor can get ready to get proper readings and then pulls data, maybe waits 30 sec to get another data sample, and another sample after 30 sec and then sleeps (fan off) and the ESP8266 goes to deep sleep, too. 

How could I realize this properly without using delay()? 

Many thanks in advance!

fu-hsi, 2017-10-06:
This method is independent from used library:

1. Wake up the main processor (ESP).
2. Wake up the PMS sensor.
3. Sleep the main processor for about 30 sec.*.
4. After wake up, take couple measurements in active mode (interval is about 2.3 sec.  if the concentration change is small) and calculate average.
5. Sleep the PMS sensor.
6. Sleep the main processor for 10 min.
7. Go to step 1 :-)

* Energy savings are negligible given the current consumption of the fan.

maddhin, 2017-10-09:
yeah, that is what I want to do but I could not get it going - but as noob it is rather confusing to understand how to deal with the PMS7003. You don't happen to have a sketch you could share? :) Then I can work/learn through that. 
I'm using this library (and PMS7003) on a ESP8266/Wemos. But I'm flexible in turns of library as long as it works;)) So far I haven't found another one for ESP8266. 
Many thanks in advance!!!

fu-hsi, 2017-10-09:
Just read this: https://learn.sparkfun.com/tutorials/esp8266-thing-hookup-guide/example-sketch-goodnight-thing-sleep-mode
> Notice there’s nothing in the loop(). The program halts when the ESP.deepSleep(30000000) is called. After 30 seconds, when the ESP8266 wakes up, it’ll start running code back at the beginning of setup().

The most important thing is to recognize the state of the sensor when ESP is activated.
For this, use the eeprom memory, save sensor sleep state and then read or wake up the sensor.

maddhin, 2017-10-09:
The ESP8266 deep sleep is no problem, I can do that but I struggle with handling the PMS;) 

So basically your „most important thing“ is exactly the stuff I fail to manage;) 

You see in the sketch I’m using active, passive mode etc and delays but it’s barely working and if the delays are higher, the PMS is not responding at all - so with this sketch I can’t even measure in 5min Intervalls :)) 
I also tried to do this with the manual/info offered by the library but my skills are not enough yet to fully understand the states etc. 

Any help is highly appreciated!!

maddhin, 2017-10-09:
Ah, how rude! @Fu-hsi: many thanks for your help!!!

maddhin, 2017-10-09:
Ah, how rude! I forgot: @fu-hsi: many thanks for your help!!!

fu-hsi, 2017-10-09:
If you have problems with library, you can try mine:
https://github.com/fu-hsi/PMS

maddhin, 2017-10-09:
Thanks, that libary looks less complicated - will give it a try!

closed 2018-04-23 by jbanaszczyk