jbanaszczyk/pms5003 #35: Question about sleep and wakeup

Belphemur, 2022-01-28:
Hello,

I went through the spec sheet of the PMS5003 and I have a couple of question about sleep/wake that I couldn't find in the specsheet neither the lib documentation and I hope you can help me with them.

## Question
1. Does the WakeUp/Sleep works in active mode or only in passive mode ?
2. Is the PIN 3 (SET) needed for that ? Or does the commands goes through the normal serial port ?

closed 2022-01-28 by Belphemur

jbanaszczyk, 2022-01-28:
Hello. As i remember (I don't own any PMS5003):
1. Sleep/WakeUp works both in active and passive mode
2. (for sure) PIN3 is not needed. You can send CMD_SLEEP = __uint24{ 0x0000e4 } **or** use PIN 3 - any of them

Belphemur, 2022-01-30:
@jbanaszczyk Thanks for those infos :)